seemecnc/Firmware #3: Retore Wantai Motor Comment in configuration.h

loadedsith, 2016-09-25:
https://github.com/seemecnc/Repetier-091-ROSTOCKMAX/blob/master/Repetier-091-ROSTOCKMAX/Repetier/Configuration.h#L701 

Says: 
    #define MOTOR_CURRENT {140,140,140,130,0} // Values 0-255 (RAMBO 135 = ~0.75A, 185 = ~1A)  use 140 for xyz and 130 for the E if using Kysan motors and 175 xyz and 200 if using wantai motors

Where the new version just says

    #define MOTOR_CURRENT {140,140,140,130,0} // Values 0-255 (RAMBO 135 = ~0.75A, 185 = ~1A)  

The extra tip here was very helpful after my new hot end firmware update started giving me issues in the extruder motor, could we restore this comment, and possibly add a reminder to the firmware updating docs?

With lots of love!
-g-

loadedsith, 2016-09-26:
is dupe of #4

johnoly99, 2016-10-12:
Good point loadedsith,

Added comment back into configuration.h for motor current defs.  See commit #4

closed 2016-10-12 by johnoly99