seemecnc/Firmware #9: Auto-leveling is still not working right in the dev branch.

wwendorf, 2016-11-19:
Basically, I'm getting adhesion on the right side of the printer (from x0), but on the left side, no dice.

I posted some information in the forum on the bed leveling issue, some numbers and results that I have gotten from testing.

Would really like to get this fixed so I can get some printing done.

DerStig, 2016-11-22:
I am having the same problems nothing 20mm from the center will adhere on the X tower side. Prints are also very smashed between the y and Z towers.  End stops are adjusted all to read 0 or 1 

While I can print a Benchy VERY well, I can't print anything bigger

wwendorf, 2016-11-22:
I'm having the same problem.  Contacted seemecnc support today, and they wanted me to print this file (http://repables.com/r/800/) and do an eeprom export using matter control, and open a ticket.

I have done that, as well as provided a video showing the problems with the calibration print.

johnoly99, 2016-11-24:
There's been a lot of commits to the dev branch in the past few days.  Testing at the shop has shown much better results.  Re-pull and try it now

closed 2016-11-24 by johnoly99